HKUST-Aerial-Robotics/GVINS #3: time-synchronized logic with UTC

DarrenWong, 2021-04-15:
Dear Officer,

Thank you for sharing your great job. I wanna ask about the time-synchronized logic here.
https://github.com/HKUST-Aerial-Robotics/GVINS/blob/c3885792c55f996f78807445fa2707f64af51d84/estimator/src/estimator_node.cpp#L274

Here: 
time_diff_gnss_visensor = next_pulse_time - trigger_msg->header.stamp.toSec() + LEAP_SECONDS; 

next_pulse_time: pps GPS time in UTC format， remove the leap seconds in epoch2time() already? Correct me if I am wrong
trigger_msg->header.stamp.toSec(): VI device's UTC time
LEAP_SECONDS ： 18seconds

As the system is time_sync and gps_time = utc_time + LEAP_SECONDS  according to here.http://leapsecond.com/java/gpsclock.htm
Why do you need to add the LEAP_SECONDS to calculate the time_diff? 

Print the time difference of time_diff_gnss_visensor is quite large. I am running with data2, complex_environment.bag
[image]

shaozu, 2021-04-16:
Hi, @DarrenWong. You're right on the definition of `next_pulse_time` and `trigger_msg->header.stamp.toSec()`, except that the `next_pulse_time` is actually not converted by `epoch2time` function and `epoch2time` does not consider leap seconds at all. 

We use the variable `time_diff_gnss_visensor` to describe the time difference between the **GNSS time** and **device local time**. Since the PPS time is reported in UTC time base under u-blox default setting, we add LEAP_SECONDS to `next_pulse_time` to obtain the corresponding GNSS time. Then we calculate `time_diff_gnss_visensor` by:
`time_diff_gnss_visensor = (next_pulse_time + LEAP_SECONDS) - trigger_msg->header.stamp.toSec();`.

DarrenWong, 2021-04-16:
I see. Thank you for your explanation.

closed 2021-04-16 by DarrenWong

shaozu, 2021-04-16:
You're welcome. Actually we have update our code so it should be more clear now. If you're interested, please check it out [here](https://github.com/HKUST-Aerial-Robotics/GVINS/blob/fae39f202810ab2b44d93a754b00f099419cc746/estimator/src/estimator_node.cpp#L264). 

Note that to run the new version of code, you may need update the dataset as we changed some topic names and data types inside.

DarrenWong, 2021-04-17:
Great~ I will check it out later~ 
The docker is a quick way to test your repo, and as some people might running under Ubuntu 18.04 (included me), you can try to change the  __$ROS_DISTRO__ to  __kinetic__ directly, and change the __ros:gvins__ to __gvins:latest__ or change your Makefile. I can submit a quick PR to fix it if you want to support 18.04 with docker
https://github.com/HKUST-Aerial-Robotics/GVINS/blob/b7a16eb0192b48b2b8d140be3dfaf111fe204f16/docker/run.sh#L47
https://github.com/HKUST-Aerial-Robotics/GVINS/blob/b7a16eb0192b48b2b8d140be3dfaf111fe204f16/docker/run.sh#L42

and minor revise the

shaozu, 2021-04-18:
Thanks for your advice. The docker tag for GVINS is already updated. As for the **$ROS_DISTRO** variable, I guess it will be replaced to **kinetic** within the docker container. I just test that on another 18.04 machine, and it works well. If you encounter any further problem when run docker with Ubuntu 18.04, please feel free to let me know.

DarrenWong, 2021-04-18:
Might be just occurred on my desktop, a capture is shown below. Never mind~ if anyone occurs similar issue should be easier to find the solution, then replace it. 
Again, thank you for your great work and sharing!

[image]

shaozu, 2021-04-18:
OK. ^_^