HKUST-Aerial-Robotics/GVINS #5: Could not find the required component 'gnss_comm'.

jingshaojing, 2021-04-17:
system:Ubuntu 18.04.4
eigen:3.3.4
ceres:1.14
(but there is no error in build gnss_comm)

I have build the gnss_comm package as instruction 

```
mkdir - p gnss_comm_ws/src
cd gnss_comm_ws/src 
git clone https://github.com/HKUST-Aerial-Robotics/gnss_comm.git
cd ..
catkin_make
source ./devel/setup.bash
```
there is no error 
then  I build GVINS in another workspace as follows 
```
mkdir -p GVINS_ws/src
cd GVINS_ws/src
git clone https://github.com/HKUST-Aerial-Robotics/GVINS.git
cd ..
catkin_make
```
the error is:
```
Could not find the required component 'gnss_comm'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "gnss_comm" with
  any of the following names:

    gnss_commConfig.cmake
    gnss_comm-config.cmake

  Add the installation prefix of "gnss_comm" to CMAKE_PREFIX_PATH or set
  "gnss_comm_DIR" to a directory containing one of the above files.  If
  "gnss_comm" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  GVINS/estimator/CMakeLists.txt:8 (find_package)

```
so if I have to give path in right path? or other method?

shaozu, 2021-04-17:
Hi, @jingshaojing. You may need to put those two repositories into one catkin workspace.

jingshaojing, 2021-04-17:
I  guess so ，and it really works! thanks, but   I  have another question , the bule /gnss_enu_path and the green /gvins/path is  not coincident ?
[image]

shaozu, 2021-04-17:
The green path is in local frame, while the blue in ENU frame. There is a yaw offset between these two frames.

closed 2021-04-19 by shaozu

jingshaojing, 2021-04-19:
thanks for your work。 @shaozu