HKUST-Aerial-Robotics/GVINS #8: Issues when enabling gnss

airuchen, 2021-07-02:
Hi, thanks for the nice work. The demo sports_field really shows perfect performance. 

### 1. What's the necessary condition to generate the global path?
 The local gvins/path can be generated with RealSense D435i without gnss. But after piping the gnss topics correctly (also using ublox_driver from your repo), the altitude and longitude information do not show up and gvins/enu_path is not generated.  The yaml setup for gnss part is the same as provided file.

### 2. The following parameters do not show up in the code. What's the function of it?
```
gnss_iono_default_parameters: !!opencv-matrix   # unused parameters
   rows: 1
   cols: 8
   dt: d
   data: [0.1118E-07,  0.2235E-07, -0.4172E-06,  0.6557E-06,
          0.1249E+06, -0.4424E+06,  0.1507E+07, -0.2621E+06]
```

shaozu, 2021-07-03:
Hey, @airuchen. Thanks for your interests on this project.

For your first question, I guess the hardware synchronization is not handled on your platform. But if you are confident with your system time, you can simply set the `gnss_local_online_sync` to 0 in your config file to skip that.

For the `gnss_iono_default_parameters` parameter, it is used to serve as the default ionospheric parameters if the broadcast one is not received yet. I supposed to modify the code after renaming the parameter but somehow forgot. That part has just been fixed and thanks for pointing it out.

airuchen, 2021-07-15:
Thanks for the answer. 

Eventually, I was able to obtain the global path by tuning down the threshold of gnss_meas_buf in bool: Estimator::GNSSVIAlign(). Could you give me some hint why the threshold is set to 10? And what if the buffer amount is not enough?

shaozu, 2021-07-18:
Hi, @airuchen. Having 10 satellites tracked usually means that we get measurements from at least 2 constellations. In that case the bias brought by a single constellation can be reduced in the initialization phase. 

Actually 4 satellites from a single constellation are enough for initialization, and if later a new constellation comes in during the optimization phase you may (or may not, depending on the constellation combination) notice some small shift on the global position estimation caused by the unmodeled constellation bias.

airuchen, 2021-07-18:
Thanks again. All clear.

closed 2021-07-18 by airuchen

SSZ1, 2022-11-04:
> Thanks for the answer.
> 
> Eventually, I was able to obtain the global path by tuning down the threshold of gnss_meas_buf in bool: Estimator::GNSSVIAlign(). Could you give me some hint why the threshold is set to 10? And what if the buffer amount is not enough?

Hi~Yuwen, sorry to bother you, but I have similar problem.  You said you didn’t get gvins/enu_path and finally obtained it by tuning down the threshold of gnss_meas_buf, which is the minim numbers of the satelites. I did this too. I set the value to 6 and it worked. It showed the path in the rviz. But when I echo the /gvins/enu_pose. It showed nothing.  I’m wondering if you ever met this question.And could you kindly offer some advice for me? Thanks a lot for your time.