HKUST-Aerial-Robotics/GVINS #11: Compilation issues due to Ceres version 2.0.0

GSORF, 2021-08-04:
Hello everyone,

I am getting compiler errors when Ceres functions are getting called. I suppose this is due to a version conflict. The README states that version 1.12.0 is used. However, the official instructions on http://ceres-solver.org/installation.html will install version 2.0.0. Is this the cause of my issue? In that case I would like to contribute by updating the README with clear instructions on how to install this specific version of Ceres.

Here is my terminal output which documents the compilation error:

```
[...]
Scanning dependencies of target gvins
[ 44%] Building CXX object GVINS/estimator/CMakeFiles/gvins.dir/src/estimator_node.cpp.o
In file included from /usr/local/include/ceres/internal/parameter_dims.h:37,
                 from /usr/local/include/ceres/internal/autodiff.h:151,
                 from /usr/local/include/ceres/autodiff_cost_function.h:130,
                 from /usr/local/include/ceres/ceres.h:37,
                 from /home/akp/catkin_ws/src/GVINS/estimator/src/initial/initial_sfm.h:2,
                 from /home/akp/catkin_ws/src/GVINS/estimator/src/estimator.h:9,
                 from /home/akp/catkin_ws/src/GVINS/estimator/src/estimator_node.cpp:15:
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:64:21: error: ‘integer_sequence’ is not a member of ‘std’
   64 | struct SumImpl<std::integer_sequence<T, N, Ns...>> {
      |                     ^~~~~~~~~~~~~~~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:64:21: error: ‘integer_sequence’ is not a member of ‘std’
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:64:46: error: wrong number of template arguments (3, should be 1)
   64 | struct SumImpl<std::integer_sequence<T, N, Ns...>> {
      |                                              ^~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:60:8: note: provided for ‘template<class Seq> struct ceres::internal::SumImpl’
   60 | struct SumImpl;
      |        ^~~~~~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:64:49: error: expected unqualified-id before ‘>’ token
   64 | struct SumImpl<std::integer_sequence<T, N, Ns...>> {
      |                                                 ^~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:71:21: error: ‘integer_sequence’ is not a member of ‘std’
   71 | struct SumImpl<std::integer_sequence<T, N1, N2, Ns...>> {
      |                     ^~~~~~~~~~~~~~~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:71:21: error: ‘integer_sequence’ is not a member of ‘std’
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:71:51: error: wrong number of template arguments (4, should be 1)
   71 | struct SumImpl<std::integer_sequence<T, N1, N2, Ns...>> {
      |                                                   ^~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:60:8: note: provided for ‘template<class Seq> struct ceres::internal::SumImpl’
   60 | struct SumImpl;
      |        ^~~~~~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:71:54: error: expected unqualified-id before ‘>’ token
   71 | struct SumImpl<std::integer_sequence<T, N1, N2, Ns...>> {
      |                                                      ^~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:78:21: error: ‘integer_sequence’ is not a member of ‘std’
   78 | struct SumImpl<std::integer_sequence<T, N1, N2, N3, N4, Ns...>> {
      |                     ^~~~~~~~~~~~~~~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:78:21: error: ‘integer_sequence’ is not a member of ‘std’
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:78:59: error: wrong number of template arguments (6, should be 1)
   78 | struct SumImpl<std::integer_sequence<T, N1, N2, N3, N4, Ns...>> {
      |                                                           ^~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:60:8: note: provided for ‘template<class Seq> struct ceres::internal::SumImpl’
   60 | struct SumImpl;
      |        ^~~~~~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:78:62: error: expected unqualified-id before ‘>’ token
   78 | struct SumImpl<std::integer_sequence<T, N1, N2, N3, N4, Ns...>> {
      |                                                              ^~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:85:21: error: ‘integer_sequence’ is not a member of ‘std’
   85 | struct SumImpl<std::integer_sequence<T, N>> {
      |                     ^~~~~~~~~~~~~~~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:85:21: error: ‘integer_sequence’ is not a member of ‘std’
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:85:41: error: wrong number of template arguments (2, should be 1)
   85 | struct SumImpl<std::integer_sequence<T, N>> {
      |                                         ^
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:60:8: note: provided for ‘template<class Seq> struct ceres::internal::SumImpl’
   60 | struct SumImpl;
      |        ^~~~~~~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:85:42: error: expected unqualified-id before ‘>’ token
   85 | struct SumImpl<std::integer_sequence<T, N>> {
      |                                          ^~
/usr/local/include/ceres/internal/integer_sequence_algorithm.h:91:21: error: ‘integer_sequence’ is not a member of ‘std’
[... 547 lines cut ...]
/usr/local/include/ceres/internal/autodiff.h:357:39: error: no type named ‘Parameters’ in ‘class ceres::internal::ParameterDims<false, 4, 3, 3>’
  357 |   Take1stOrderParts<Parameters>::Apply(
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  358 |       num_outputs, residuals_as_jets.data(), jacobians);
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/ceres/internal/autodiff.h:152,
                 from /usr/local/include/ceres/autodiff_cost_function.h:130,
                 from /usr/local/include/ceres/ceres.h:37,
                 from /home/akp/catkin_ws/src/GVINS/estimator/src/initial/initial_sfm.h:2,
                 from /home/akp/catkin_ws/src/GVINS/estimator/src/estimator.h:9,
                 from /home/akp/catkin_ws/src/GVINS/estimator/src/estimator_node.cpp:15:
/usr/local/include/ceres/internal/variadic_evaluate.h: In instantiation of ‘bool ceres::internal::VariadicEvaluateImpl(const Functor&, const T* const*, T*, const void*) [with ParameterDims = ceres::internal::ParameterDims<false, 4, 3, 3>; Functor = ReprojectionError3D; T = double]’:
/usr/local/include/ceres/internal/variadic_evaluate.h:108:45:   required from ‘bool ceres::internal::VariadicEvaluate(const Functor&, const T* const*, T*) [with ParameterDims = ceres::internal::ParameterDims<false, 4, 3, 3>; Functor = ReprojectionError3D; T = double]’
/usr/local/include/ceres/autodiff_cost_function.h:207:55:   required from ‘bool ceres::AutoDiffCostFunction<CostFunctor, kNumResiduals, Ns>::Evaluate(const double* const*, double*, double**) const [with CostFunctor = ReprojectionError3D; int kNumResiduals = 2; int ...Ns = {4, 3, 3}]’
/usr/local/include/ceres/autodiff_cost_function.h:200:8:   required from here
/usr/local/include/ceres/internal/variadic_evaluate.h:80:65: error: ‘ParameterBlockIndices’ was not declared in this scope
   80 |       functor, input, output, IsDynamic(), ParameterBlockIndices());
      |                                            ~~~~~~~~~~~~~~~~~~~~~^~
make[2]: *** [GVINS/estimator/CMakeFiles/gvins.dir/build.make:63: GVINS/estimator/CMakeFiles/gvins.dir/src/estimator_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3720: GVINS/estimator/CMakeFiles/gvins.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j1 -l1" failed
```

Thank you for your time

GSORF, 2021-08-04:
Ok, I have tested it with Ceres version 1.12.0 and it compiles.

I have proposed a minor modification of the README to make this more explicit to the users, see #12 .

Now I have to cope with OpenCV issues, which is probably related to a version difference again. But I think this will be another easy fix.

Kind regards.

shaozu, 2021-09-11:
Hi, @GSORF. Thanks for your efforts on this issue. I will leave this issue open in case someone has the same problem. 

I just want to add that another solution to this problem is to build and test the whole project using docker, as shown [here](https://github.com/HKUST-Aerial-Robotics/GVINS#3-docker-support). By so the GVINS environment is totally isolated from your local one and everything should go smoothly.

Kranthi18reddy, 2022-01-28:
> Hello everyone,
>
> I am getting compiler errors when Ceres functions are getting called. I suppose this is due to a version conflict. The README states that version 1.12.0 is used. However, the [...]

Using CXX flags as "std=c++14" will compile Ceres 2.0.0