HKUST-Aerial-Robotics/GVINS #16: Stereo Compatibility

ardabbour, 2021-09-09:
Hello and thank you for a great piece of work! From what I gather, there is nothing stopping this method working with a pair of stereo cameras (such as the RS D455); I was just wondering what the changes needed to be done in the codebase would be and if you are looking forward to doing it

shaozu, 2021-09-11:
Hi, @ardabbour. Yes, stereo camera setting can definitively improve the overall performance of the estimator. However, in current stage we only investigate the limit of the estimator in minimum sensor setting, i.e., only monocular+IMU+GNSS. If you are interested in the stereo setup, you may refer to the [VINS-Fusion](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion) for how to build visual constraints from two cameras.

closed 2021-09-14 by ardabbour

sunbingfeng, 2022-05-30:
@ardabbour, 

Hi, how's your final performance using stereo cameras?
Thanks!

ardabbour, 2022-05-31:
@sunbingfeng We decided to use a different library, so did not progress on that front.