HKUST-Aerial-Robotics/GVINS #20: IMU noise unit

Weirdo-p, 2021-12-01:
I am confused to the units of IMU parameters when I want to run GVINS with my own dataset. Would you mind if you could describe them thoroughly.
```
#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.08          # accelerometer measurement noise standard deviation. #0.2   0.04
gyr_n: 0.004         # gyroscope measurement noise standard deviation.     #0.05  0.004
acc_w: 0.00004         # accelerometer bias random work noise standard deviation.  #0.02
gyr_w: 2.0e-6       # gyroscope bias random work noise standard deviation.     #4.0e-5
```

shaozu, 2021-12-06:
Hi, @Weirdo-p. You may refer to the page 14 and 15 of [this technical report](http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf) for the units of the IMU parameters.

Weirdo-p, 2022-01-02:
@shaozu Thank you. But I am still confused. I have imu parameters (NovAtel i300) which are tested in loose coupled GNSS/INS with the same units that described in that technical report. But the result of my own dataset is bad especially in z-axis in a feature-enough scene. Have you ever tested i300 before? Do you have any suggestions?

shaozu, 2022-01-04:
I haven't tested the NovAtel i300, but you may need to calibrate it using [imu_utils](https://github.com/gaowenliang/imu_utils).

closed 2022-01-26 by shaozu