HKUST-Aerial-Robotics/GVINS #21: How can I get time of world frame?

Weirdo-p, 2021-12-04:
Hi, thanks for your great work.
I want to evaluate accuracy of GVINS(without GNSS) which means I need time of reference frame (or world frame). How can I get this?
Thanks in advance.

shaozu, 2021-12-06:
Hi, @Weirdo-p. If you have an RTK as your ground truth, and you need to align the RTK time with the local time of GVINS (no GNSS), you can use NTP to sync your equipment to UTC time and simply add leaps seconds between UTC and GPS Time (currently 18 seconds). An accuracy of tens of milliseconds can be achieved if the network is good.

Weirdo-p, 2021-12-10:
thank  u~
And one more question. What is the difference between LOCAL_SIZE and GLOBAL_SIZE? And why you used different sizes in marginalization.
thank you again

shaozu, 2021-12-13:
You're welcome.

The `LOCAL_SIZE` and `GLOBAL_SIZE` are the concepts of local parameterization of the [ceres](http://ceres-solver.org/index.html) library. You may refer [here](http://ceres-solver.org/nnls_modeling.html#localparameterization) for the details. Essentially, the reason for using different sizes is to avoid the singularity issue when evaluate the jacobian matrix.

closed 2021-12-25 by shaozu