HKUST-Aerial-Robotics/GVINS #22: unit2rcv_pos in code

dean1314, 2021-12-08:
Hello
I find you use dopp to constrain the position in code as follow
Eigen::Matrix3d unit2rcv_pos;
for (size_t i = 0; i < 3; ++i)
{
    for (size_t j = 0; j < 3; ++j)
    {
        if (i == j)
            unit2rcv_pos(i, j) = (norm2-rcv2sat_ecef(i)*rcv2sat_ecef(i))/norm3;
        else
            unit2rcv_pos(i, j) = (-rcv2sat_ecef(i)*rcv2sat_ecef(j))/norm3;
    }
}
but this method don't mention it in your paper, so can you give me some papers about this method

shaozu, 2021-12-13:
Hi, @dean1314. the term `unit2rcv_pos` is a part of the jacobian of the Doppler residual with respect to the receiver's position. It does not involve any special method but just plain derivatives. In fact, that term is very small, and you can safely ignore it.

dean1314, 2021-12-15:
Thanks
but i don't see any receiver's position in the Doppler residual format, the Doppler residual format as follow
[image]

shaozu, 2021-12-15:
It's in the directional vector \kappa.

closed 2021-12-25 by shaozu