HKUST-Aerial-Robotics/GVINS #23: How to dispaly the maps

Abner0907, 2021-12-14:
@shaozu  Hello, I am very like this project. it is my needed. I have run it in my pc. all works good. Thank you very much for you nice work.

I have a simple question:
I want know how to display this path trajectory/path in the maps as yours.what tools should i use. how to configure it?  I am very appreciate that you can give some reference. thank you very much again. I can display the path in rviz without any map backgrounds.
[image]

shaozu, 2021-12-14:
Hi, @Abner0907. I used the javascript API provided by Google Maps, and you may want to refer [this page](https://developers.google.com/maps/documentation/javascript/overview) for some instructions.

Abner0907, 2021-12-14:
@shaozu  Thank you very much, I will have a try now.

Abner0907, 2021-12-17:
closed as now issues now

closed 2021-12-17 by Abner0907