HKUST-Aerial-Robotics/GVINS #25: Question about UBLOX Config

Abner0907, 2021-12-17:
hello, I have follow your steps, do the VI slam, The result is below, looks good. 
[image]

I have two question, hopes your suggestions:
question 1: I have a f9p module, but I have not rtcm data, so it will never get rtk fix. Can this data fuse with the image&imu data?
question 2: I use my f9p module, run the command, `roslaunch ublox_driver ublox_driver.launch`, I can see the topic show as below: 
>/ublox_driver/ephem
/ublox_driver/glo_ephem
/ublox_driver/iono_params
/ublox_driver/range_meas
/ublox_driver/receiver_lla
/ublox_driver/receiver_pvt
/ublox_driver/time_pulse_info

How ever, when I `rostopic echo /ublox_driver/receiver_lla`, no topic is published/display. Do I need some special config through `u-center` or some other tools?

shaozu, 2021-12-17:
Hi, @Abner0907. For your questions,
1. You don't need RTK solutions for GVINS. The RTK is just used as ground truth for our experiments evaluation.
2. Yes, you need to config your receiver through `u-center` first. [Here](https://github.com/HKUST-Aerial-Robotics/ublox_driver/blob/main/config/ucenter_config_f9p_gvins.txt) you can find the config that we used.

Abner0907, 2021-12-17:
@shaozu  Hi, thank you for the reply. 
1: I have follow your info [commit](https://github.com/HKUST-Aerial-Robotics/ublox_driver/commit/5de78a593bbf909a3bf30eb049561e4f62f1e163) and config the two message. `UBX-RXM-RAWX`, and `UBX-NAV-PVT`.
 2: However, I can not config the `UBX-RXM-SFRBX`  through u-center. Does it must? and is it enough for the gvins fusion.

Abner0907, 2021-12-22:
closed, As I have used an private f9p protocol, it parse the data not correct, which cause this issues.

closed 2021-12-22 by Abner0907

ZBY-creator, 2022-08-26:
> closed, As I have used an private f9p protocol, it parse the data not correct, which cause this issues.

Did you meet the problem that the config file used by GVINS cannot saved by your own chip?