HKUST-Aerial-Robotics/GVINS #26: gnss_enable is enabled, can not get global path

Abner0907, 2021-12-18:
@shaozu  Hello, disturb to again.
I have a question about how to enable `gnss_enable`.

as you can see #25. If I set `gnss_enable:0`, it can get the nice local position.
however, when I set  `gnss_enable:1`, it can not get the gloabl position.  So i start to debug and get some questions. hope your reply.
**question:**
**1:** I found that I have no  topic `/external_trigger` is published. how does it work? or is it must?
**2:** what 's the function of  the parameter `gnss_local_time_diff` ?
**3:** Is there some reference for explanation `gnss_iono_default_parameters`.  sorry, I have not found any reference. Can it be changed? what it depends on?
**4:**  In `GNSSVIAlign()` function, I found it always return false as the reason of `gnss_meas_buf[i].empty() -> false` && `gnss_meas_buf[i].size() = 0`.
https://github.com/HKUST-Aerial-Robotics/GVINS/blob/d2cf40b49c6eb0e6ad3caa4f613713983be3fd74/estimator/src/estimator.cpp#L566-L570
But I can confirm that it has push the data in the buffer here
https://github.com/HKUST-Aerial-Robotics/GVINS/blob/d2cf40b49c6eb0e6ad3caa4f613713983be3fd74/estimator/src/estimator.cpp#L326-L327.
Very confused me. 

Thank you for sharing your great job.

Abner0907, 2021-12-18:
@shaozu This is my monitor topic bandwidth.
[image]

Abner0907, 2021-12-22:
closed, As I have used an private f9p protocol, it parse the data not correct, which cause this issues.

closed 2021-12-22 by Abner0907

maghauke, 2022-05-01:
@Abner0907 I have a similar problem, could you specify what you mean about a “private f9p protocol”? Do you mean your setup of the ublox-system or do you mean the GVINS-ublox-driver?

Abner0907, 2022-06-02:
@maghauke  I have ref the rtklib drivers and do some changes in the file of `gnss_ubox_driver/src/ublox_message_processor.cpp`, which now all work good for me.
as the private f9p, I do not think so, it is my not correct words.

SSZ1, 2022-11-06:
> 

Hi~Bro. So have you figured out the meaning of /external_trigger ?  What's its function?? And how could we get it?