HKUST-Aerial-Robotics/GVINS #27: Question about the "uura" in gnss_psr_dopp_factor.cpp

bryantaoli, 2021-12-19:
[image]
What does "uura" stand for and why is it calculated like this？
Thank you.

shaozu, 2021-12-25:
The `uura` has no special meaning and it is just a part of the information weight associated with this factor. 
The information weight itself is an empirical formula as we verified in a long-term static receiver experiment.

closed 2022-01-04 by shaozu

bryantaoli, 2022-02-17:
[image]
In your GVINS' paper, you said the clock jumps were avoided or carefully handled. But in your code, I cannot find anything about handling clock jump, could you please tell me how to  avoid or handle clock jumps in GVINS, thank you so much！