HKUST-Aerial-Robotics/GVINS #34: Issuses on the Jacobian of the GnssPsrDoppFactor.

Nronaldo, 2022-05-25:
Dear authors:
Could you please tell me how to calculate the doppler factor Jacobian with respect to the position state? I do not understand the codes that are marked by a red cycle in the following figure.  I refer to the arxiv version of your article and do not find the details about this Jacobian.  Could you give me a guide or reference? Thank you!
[image]

LSXiang, 2022-06-24:
@Nronaldo I have the same problem. Do you understand now?

Nronaldo, 2022-06-24:
> @Nronaldo I have the same problem. Do you understand now?

I am sorry I do not.. If you know ,please tell me.

Mil1ium, 2022-09-07:
Maybe this could help you. @Nronaldo @LSXiang 
I'm not necessarrily right, just for reference.

[image]

[image]

[image]

LSXiang, 2022-09-09:
> Maybe this could help you. @Nronaldo @LSXiang I'm not necessarrily right, just for reference.
> 
> [image]
> 
> [image]
> 
> [image]

@Mil1ium Great, thanks a lot.

LSXiang, 2022-09-09:
@Mil1ium Great, thanks a lot.

Abner0907, 2022-12-09:
@Mil1ium HI, 
Thanks for your notes above [here](https://github.com/HKUST-Aerial-Robotics/GVINS/issues/34#issuecomment-1239708167). Can you share some papers or references links about the algorithm.

aphasiayc, 2024-08-29:
I have a related question. Both in GnssPsrDoppFactor and in psr_pos (which i believe is largely inherited from RTKLIB), the derivatives of the Sagnac term do no appear in the Jacobians. Why is that?