HKUST-Aerial-Robotics/GVINS #36: About the Earth's rotation

yinzhang809, 2022-06-06:
The GVINS paper and code are very clear and helpful. I have a small question about the code.

When reading 'gnss_psr_dopp_factor.cpp', I cannot find code segment that corresponds to the Earth's rotation term of formulas (16) and (20) of the GVINS paper. Is this part of the processing omitted from the code? And have you tested how much the Earth's rotation affects accuracy?

Looking forward to your reply!

Mil1ium, 2022-10-05:
Hi, @yinzhang809 
I think the earth's rotation compensation may be included in Sagnac effect correction.

nwpuclydewang, 2023-10-17:
I'am confused about the Sagnac effect correction in doppler estimated.Could you tell me how to explain this equation？

[image]