HKUST-Aerial-Robotics/GVINS #44: Questions about running GVINS with my own device.

SSZ1, 2022-10-25:
Hi~bro. I ran the GVINS with my own device (including a realsense D455 as the VIS and a [H-RTK-F9P](http://www.holybro.com/product/h-rtk-f9p/) as the GNSS module ).  I set all the device right. And the GPS runs fine. But it shows "length error" in the Ublox-driver terminal windiow.But when I run the GVINS project with the estimate_extrinsic 2. But After I roslaunched the GVINS , the terminal windows shows only  "init begins". And I opened the rviz , it had the original image and the feature point image but no pose. I echoed the enu_pose and other messages outputed by GVINS, they are all none. So why is this? 

[image]

[image]

    Could you give me some advice about this? Thanks a lot.

ZoYooJy, 2022-12-27:
could u tell the details about the Synchronization between vi & gnss receiver module? thx very much!!!

SaiSugunSegu, 2023-06-26:
@ZoYooJy , what method of Synchronization would you suggest for synchronizing GNSS and VI 

I use GPIO trigger to synchronize the VI system, 

- would you prefer any other good method for Synchronization - be it (Camera + IMU) or (GNSS+ camera+ IMU).

what do you do if the sensor doesn't have GPIO pins?