HKUST-Aerial-Robotics/GVINS #46: How to get standard deviation of gnss receiver clock bias and clock drift rate?

Mil1ium, 2022-11-14:
Thanks for your nice work. @shaozu 

I've got a new question about GNSS receiver dt and ddt factor.

In your code, standard deviation of ddt is set to 0.1(unit: m/s ?) and the infomation matrix of dt is set to 50(unit: m ?).
Could you please tell me why set them to these values?

SSZ1, 2022-11-15:
> Thanks for your nice work. @shaozu
> 
> I've got a new question about GNSS receiver dt and ddt factor.
> 
> In your code, standard deviation of ddt is set to 0.1(unit: m/s ?) and the infomation matrix of dt is set to 50(unit: m ?). Could you please tell me why set them to these values?

Hi~bro. I met a problem here. I use my own device to test GVINS. And I found the green path( in local frame) and the blue path (in ENU frame) is not the same. And @shaozu  said in [issue#5](https://github.com/HKUST-Aerial-Robotics/GVINS/issues/5) said `The green path is in local frame, while the blue in ENU frame. There is a yaw offset between these two frames.` But in my test, it seemed there is not only a yaw offset between these two paths. It looked totally different. So do you know this is why? Thanks for your time.
[image]

SSZ1, 2022-11-15:
> > Thanks for your nice work. @shaozu
> > I've got a new question about GNSS receiver dt and ddt factor.
> > In your code, standard deviation of ddt is set to 0.1(unit: m/s ?) and the infomation matrix of dt is set to 50(unit: m ?). Could you please tell me why set them to these values?
> 
> Hi~bro. I met a problem here. I use my own device to test GVINS. And I found the green path( in local frame) and the blue path (in ENU frame) is not the same. And @shaozu in [issue#5](https://github.com/HKUST-Aerial-Robotics/GVINS/issues/5) said `The green path is in local frame, while the blue in ENU frame. There is a yaw offset between these two frames.` But in my test, it seemed there is not only a yaw offset between these two paths. It looked totally different. So do you know this is why? Thanks for your time.
 [image]
[image]

SSZ1, 2022-11-15:
> Thanks for your nice work. @shaozu
> I've got a new question about GNSS receiver dt and ddt factor.
> In your code, standard deviation of ddt is set to 0.1(unit: m/s ?) and the infomation matrix of dt [...]
One test
> > [image]
> > [image]

two test
[image]
[image]

ooww0123TW, 2024-02-13:
I think the reason for the two paths are different is varying yaw offset and anc_ecef value.