HKUST-Aerial-Robotics/GVINS #50: urban driving huge error

haibo9916, 2023-03-18:
When using your GVINS to test the urban driving dataset, I encountered a significant error on a section of the road. What is the problem and how to solve it.？The huge error occurred in the red box of the following figure
[image]
@shaojie @dvorak0 @WayneTimer @hlx1996

bryantaoli, 2023-09-27:
> When using your GVINS to test the urban driving dataset, I encountered a significant error on a section of the road. What is the problem and how to solve it.？The huge error occurred in the red box of the following figure [image] @shaojie @dvorak0 @WayneTimer @hlx1996

Try 'if(fabs(g.norm() - G.norm()) > 1.0 || s < 0)' to 'if(fabs(g.norm() - G.norm()) > 2.0 || s < 0)'. It maybe help, I try this sentence in other dataset.

ooww0123TW, 2024-02-13:
It seems that reset function has been called at that time due to failure detection.

GUAIYISHASHOU, 2025-04-07:
May I ask what dataset you are using? Is it data containing raw GNSS observations?