HKUST-Aerial-Robotics/GVINS #51: urban driving huge error

haibo9916, 2023-03-18:
When using your GVINS to test the urban driving dataset, I encountered a significant error on a section of the road. What is the problem and how to solve it.？The huge error occurred in the red box of the following figure
@shaojie @dvorak0 @WayneTimer @hlx1996 

[image]

aa1811111, 2023-09-16:
you have solved this problem now?

haibo9916, 2023-09-16:
> you have solved this problem now?
no

UCAS-apm-Bo-Zhang, 2024-05-04:
I also encountered the same issue. When trajectory reset happens, you can see warning messages output in the terminal. This is the program's built-in failureDetection() function, which does the following: a. If a fault is detected, it performs operations like clearing states, setting parameters, etc., and returns; b. If no fault is detected, it performs operations like sliding window, removing failed feature points, updating key pose vectors, and the latest rotation matrix and position vector. You can try commenting out the fault detection function to solve this issue, but whether it will affect accuracy is still debatable.

GUAIYISHASHOU, 2025-04-07:
> I also encountered the same issue. When trajectory reset happens, you can see warning messages output in the terminal. This is the program's built-in failureDetection() function, which does the following: a. If a fault is detected, it performs operations like clearing states, setting parameters, etc., and returns; b. If no fault is detected, it performs operations like sliding window, removing failed feature points, updating key pose vectors, and the latest rotation matrix and position vector. You can try commenting out the fault detection function to solve this issue, but whether it will affect accuracy is still debatable.我也遇到了同样的问题。当发生轨迹重置时，可以看到终端中输出的警告信息。这是程序内置的 failureDetection() 函数在起作用，该函数执行以下操作：a. 若检测到故障则执行状态清除、参数设置等操作后返回；b. 若未检测到故障则执行滑动窗口、剔除失效特征点、更新关键位姿向量及最新旋转矩阵与位置向量等操作。你可以尝试注释掉故障检测函数来解决该问题，但该操作是否会影响精度尚有待商榷。

May I ask what dataset you are using? Is it data containing raw GNSS observations?